SEAME-pt/2024-2025-HotWheels #12: Clang-tidy

reomelo, 2024-11-15:
`Description:` What is it? How does it work? Do we need it?

</br>

`Definition of done:` This issue is considered done when the above questions are clearly answered

t-pereira06, 2024-11-18:
## What is clang-tidy?

Clang-Tidy is a powerful static analysis and linting tool that is part of the LLVM/Clang ecosystem.
It is primarily used for improving the quality of C, C++, and Objective-C codebases by identifying potential
errors, style inconsistencies, and optimization opportunities.
Clang-Tidy is particularly popular in projects where code quality and maintainability are critical.

Benefits:

- Prevents Bugs: Detects potential runtime bugs early in the development cycle.
- Improves Code Quality: Ensures consistent style and modern practices.
- Saves Time: Automates repetitive refactoring and analysis tasks.
- Encourages Best Practices: Helps teams adopt and enforce modern C++ standards.

## How to install clang-tidy?
```bash
sudo apt update
sudo apt install clang-tidy
```

Check if it is installed
```bash
clang-tidy --version
```

## How does it work?

### Step 1: Generate a Compilation Database (compile_commands.json)

Clang-Tidy requires a compile_commands.json file to understand how each file in your project is being compiled.
This file lists all source files and the exact compiler flags used for them.

Generating compile_commands.json with CMake
- Add the following option when configuring the project:
```bash
cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON /path/to/project
```
Place the compile_commands.json file at the root of the project

### Step 2: Run Clang-Tidy
We can now run Clang-Tidy on the project.
```bash
clang-tidy path/to/file.cpp -p path/to/compile_commands.json
```

### Step 3: Integrate Clang-Tidy into Your Build Process
Using CMake's clang-tidy Integration
Add this line to CMakeLists.txt:
```bash
set(CMAKE_CXX_CLANG_TIDY "clang-tidy;-checks=modernize-*")
```

## Do we need it?
Using Clang-Tidy in your projects offers several benefits that enhance code quality, maintainability, and reliability. Here are key reasons why you should incorporate Clang-Tidy into your development workflow:

- Identify Bugs and Prevent Runtime Issues
- Static Analysis: 
Clang-Tidy performs static code analysis, allowing you to identify bugs, undefined behavior, and potential runtime errors before executing the code.
- Early Detection:
Issues like dereferencing null pointers, memory leaks, or use-after-free bugs can be caught during development, reducing the risk of costly debugging later.

### Enforce Coding Standards
- Consistent Code Style:
Clang-Tidy can enforce adherence to coding standards such as the Google C++ Style Guide, LLVM Coding Standards, or custom project-specific guidelines.
- Customizable Checks:
You can configure it to follow your project's unique coding practices, ensuring team-wide consistency.

### Modernize Legacy Code
- Adopt Modern C++:
Clang-Tidy helps modernize legacy C++ code by suggesting replacements for outdated patterns with modern C++ constructs (e.g., replacing NULL with nullptr or raw loops with range-based loops).
- Improved Safety and Performance:
By encouraging the use of modern C++ features, it improves the code's safety, readability, and performance.

### Optimize Performance
- Detect Inefficiencies:
It can identify inefficient code patterns and suggest optimizations, such as replacing redundant calculations or avoiding unnecessary copies.
- Better Resource Management:
Tools like Clang-Tidy encourage the use of RAII (Resource Acquisition Is Initialization) for better memory and resource management.

### Automate Refactoring
- Save Developer Time:
Clang-Tidy can automatically apply fixes for many issues, saving time on repetitive and tedious tasks.
- Batch Fixes:
It allows bulk refactoring across multiple files, making it practical to apply large-scale changes to your codebase.

### Maintain High Code Quality
- Continuous Integration (CI):
By integrating Clang-Tidy into your CI pipeline, you can ensure that new code commits meet quality standards and do not introduce regressions.
- Readable Code:
Clean and consistent code is easier to maintain, review, and extend, leading to fewer long-term issues.

reomelo, 2024-11-18:
Good job Tiago. This tool will be implemented as soon as possible and all team members should use it

closed 2024-11-18 by t-pereira06