SEAME-pt/2024-2025-HotWheels #20: Implement controls

reomelo, 2024-11-18:
`Description:` Implement movement keys to allow user control. Each key input generates a signal to control the wheels motors.

</br>

`Definition of done:` This issue is considered done when the program accepts input and sends signals to the car that interact with the wheels motors, changing values in real time.

t-pereira06, 2024-11-29:
Controls are working right now, but there's a problem where the car sometimes reboots itself and also it stops recognizing the controller.
This was tested with the old version of the OS, we are going to test on this new one.

t-pereira06, 2024-11-29:
In the new version, the OS does not reboot itself but it still stops recognizing the controller.

reomelo, 2024-12-02:
The direction controls still shutdown the car. This issue will be moved to next sprint

reomelo, 2024-12-16:
Finished in #51

closed 2024-12-16 by reomelo