SEAME-pt/2024-2025-HotWheels #27: Research speed sensor details

reomelo, 2024-11-19:
`Description:` Study about the LM 393 speed sensor. What signal does it use? How do we connect it to the ARDUINO?

</br>

`Definition of done:` This issue is considered as done when the above questions are clearly answered

Fle-bihh, 2024-11-20:
Speed Sensor Documentation

    Voltage: 3.3V ~ 5V
    Usage: Dynamo speed detecting, pulse counting, etc.
    Output:
        Digital switch output (0 and 1)
        Analog output for sensitivity

Functionality:

    Operates using infrared light emitted from the LED and received by the photo-transitor.
    As the wheel spins, the light beam is alternately blocked and unblocked:
        Block → Unblock = Pulse

Connection to Arduino:

    VCC → 5V pin on Arduino board
    GND → GND
    OUT → Digital input pin

Example: Calculating Speed

    Measure the number of pulses over a fixed time period (e.g., 1 second).
    Determine the distance covered per pulse (e.g., based on the wheel circumference).
    Calculate speed using the formula:

Speed=Time (s)Pulses per second×Distance per pulse​

Fle-bihh, 2024-11-20:
Useful link for setup with Arduino :
https://www.brainy-bits.com/post/how-to-use-a-speed-sensor-with-arduino

closed 2024-11-20 by reomelo