SEAME-pt/2024-2025-HotWheels #40: Setup Docker

Fle-bihh, 2024-12-02:
`Description:` Setup a Docker image containing every libraries needed, on Host computer, to build the Cluster Qt app. Build and run this image from the Car board.

</br>

`Definition of done:` When the Car can run the docker image from its board and display the cluster.

Fle-bihh, 2024-12-08:
Update: The binary is compiled and built from docker container (Host machine).
The binary contains all Qt libraries needed.
Only thing missing is a library to display the cluster on target  : xcb is only solution to forward display to host, but not recommended to display on target. We have to choose between linuxfb and EGLFS, make sure they are included in the binary or installed on target to run the cluster successfully.

reomelo, 2025-01-13:
A different working method of cross compilation has been implemented in Sprint 3

closed 2025-01-13 by reomelo