SEAME-pt/2024-2025-HotWheels #51: Fix car movement issue

reomelo, 2024-12-02:
`Description:` Fix the shutting down on steering issue. Is it the servo? The nano board? The scripts? Find out

</br>

`Definition of done:` This issue is considered as done when the team can maneuver the car freely without any problems or errors

t-pereira06, 2024-12-02:
Right now, the car still shuts down even with the new servo. We think there is a board issue, as we tested our SD card on another car and it worked fine. Maybe a replacement is needed.

reomelo, 2024-12-05:
We finally fixed the problem. It was indeed a hardware issue in the expansion board. After replacing it with a new one it worked perfectly. The raspberry won't be used after all. We can move on now!

closed 2024-12-05 by reomelo