SEAME-pt/2024-2025-HotWheels #79: Improve UI

Fle-bihh, 2025-01-15:
`Description:` Improve the global UI of the instrumental Cluster : wifi, battery, time placements; Circular Meters widgets looks; Circular Meters animation

</br>

`Definition of done:`This issue is done once the circular meters animation is smooth at all time (Even multiple huge speed / rpm updates), and the global aspect of the instrumental cluster looks professional and clean.

reomelo, 2025-01-17:
[image]

The new cluster has been designed. Only thing left is to implement it in the QT App

reomelo, 2025-01-23:
The new UI has been successfully implemented in the Qt App

closed 2025-01-23 by reomelo