SEAME-pt/2024-2025-HotWheels #87: Car controls integration with QT

reomelo, 2025-01-17:
`Description:` Integrate the car controls script in the QT App so we can access the controller values and communicate with the motors

</br>

`Definition of done:` This issue is considered as done when the QT App is capable of reading the inputs from the controller

Fle-bihh, 2025-01-21:
Everything is integrated in the qt app, except for the init_motors which is not working properly. (Making the car shut down due to power voltage seems to disable the motors and would permit to test the init_motors again)

Fle-bihh, 2025-01-24:
The set_motor_pwm method of EngineController was not complete. By fixing it, the class now enables the motor anytime it calls setSpeed. Car controls are now fully integrated in the qt app.

closed 2025-01-24 by Fle-bihh